gregkh/usbutils #40: Fails to build in GNOME Continuous

ebassi, 2016-04-29:
Commit 8623cb5741e149e0f4087ea01b520902190e8b90 added a C99-ism:

```
for (unsigned char i = 0; i < cap_data_len; i++) {
```

which requires -std=c99 in order to build on GCC. This was caught by the GNOME Continuous builder:

```
../lsusb.c: In function 'dump_platform_device_capability_desc':
../lsusb.c:3818:2: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
  for (unsigned char i = 0; i < cap_data_len; i++) {
  ^
../lsusb.c:3818:2: note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 to compile your code
Makefile:597: recipe for target 'lsusb-lsusb.o' failed
make[2]: *** [lsusb-lsusb.o] Error 1
make[2]: *** Waiting for unfinished jobs....
```

Full log: http://build.gnome.org/continuous/buildmaster/builds/2016/04/29/40/build/log-usbutils.txt

gregkh, 2016-04-29:
Care to provide a patch?

ebassi, 2016-04-29:
Sure; what's the preferred approach? Do you want to keep the C99-ism or do you want to keep using the older declaration-before-statement style?

closed 2016-04-29 by gregkh

tsundre, 2016-04-30:
Sorry for that. This was my first contribution ever. Didn't realize C99-isms was not ok. Is there any guidelines I should follow?

PR #41 (2016-04-29): Don't use C99-ism
Declare the loop iterator before using it.

Fixes #40
commit a0845301 (2016-04-29): Don't use C99-ism

Declare the loop iterator before using it.

Fixes #40