gregkh/usbutils #48: fails to autoreconf on ubuntu xenial

mailinglists35, 2016-12-07:
root@host:/usr/local/src# git clone https://github.com/gregkh/usbutils.git
    Cloning into 'usbutils'...
    remote: Counting objects: 1204, done.
    remote: Total 1204 (delta 0), reused 0 (delta 0), pack-reused 1204
    Receiving objects: 100% (1204/1204), 837.16 KiB | 643.00 KiB/s, done.
    Resolving deltas: 100% (732/732), done.
    Checking connectivity... done.
    root@host:/usr/local/src# cd usbutils/
    root@host:/usr/local/src/usbutils# git submodule init
    Submodule 'usbhid-dump' (https://github.com/DIGImend/usbhid-dump.git) registered for path 'usbhid-dump'
    root@host:/usr/local/src/usbutils# git submodule update
    Cloning into 'usbhid-dump'...
    remote: Counting objects: 541, done.
    remote: Total 541 (delta 0), reused 0 (delta 0), pack-reused 541
    Receiving objects: 100% (541/541), 99.04 KiB | 0 bytes/s, done.
    Resolving deltas: 100% (358/358), done.
    Checking connectivity... done.
    Submodule path 'usbhid-dump': checked out 'b18e816cbf65fe3b7f53d1d275f550c0c18e9b0f'
    root@host:/usr/local/src/usbutils# autoreconf --install --symlink
    configure.ac:47: error: possibly undefined macro: AC_PROG_LIBTOOL
          If this token and others are legitimate, please use m4_pattern_allow.
          See the Autoconf documentation.
    autoreconf: /usr/bin/autoconf failed with exit status: 1
    root@host:/usr/local/src/usbutils# apt policy autoconf
    autoconf:
      Installed: 2.69-9
      Candidate: 2.69-9
      Version table:
     *** 2.69-9 500
            500 http://ro.archive.ubuntu.com/ubuntu xenial/main amd64 Packages
            500 http://ro.archive.ubuntu.com/ubuntu xenial/main i386 Packages
            500 http://ftp.lug.ro/ubuntu xenial/main amd64 Packages
            500 http://ftp.lug.ro/ubuntu xenial/main i386 Packages
            100 /var/lib/dpkg/status

mailinglists35, 2016-12-07:
fixed by installing package `libtool`

http://superuser.com/questions/565988/autoconf-libtool-and-an-undefined-ac-prog-libtool

maybe you should add a list of requirements in the readme

mailinglists35, 2016-12-07:
i reopened just to be sure you will see it

gregkh, 2016-12-07:
autoconf and friends (like libtool) are always an implicit dependency, no need to call them out individually here.

closed 2016-12-07 by gregkh