gregkh/usbutils #49: fails to ./configure on ubuntu xenial

mailinglists35, 2016-12-07:
[...]
    checking for iconv... yes
    ./configure: line 4533: syntax error near unexpected token `LIBUSB,'
    ./configure: line 4533: `PKG_CHECK_MODULES(LIBUSB, libusb-1.0 >= 1.0.9)'

it looks for "libusb-1.0" but in ubuntu is called "libusb-1.0-0"

also 

    ./configure: line 4535: syntax error near unexpected token `UDEV,'
    ./configure: line 4535: `PKG_CHECK_MODULES(UDEV, libudev >= 196)'

ubuntu package name is libudev1

I tried replacing with ubuntu packages names but it gave the same error, then tried commenting these lines as well as usbhid-dump/configure at line 11958, but further I get this:

make[4]: Entering directory '/usr/local/src/usbutils/usbhid-dump/lib'
  CC       dev.lo
In file included from /usr/local/src/usbutils/usbhid-dump/include/uhd/dev.h:31:0,
                 from dev.c:29:
/usr/local/src/usbutils/usbhid-dump/include/uhd/libusb.h:30:20: fatal error: libusb.h: No such file or directory
compilation terminated.
Makefile:393: recipe for target 'dev.lo' failed
make[4]: *** [dev.lo] Error 1
make[4]: Leaving directory '/usr/local/src/usbutils/usbhid-dump/lib'
Makefile:422: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
make[3]: Leaving directory '/usr/local/src/usbutils/usbhid-dump'
Makefile:354: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/usr/local/src/usbutils/usbhid-dump'
Makefile:791: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/usr/local/src/usbutils'
Makefile:406: recipe for target 'all' failed
make: *** [all] Error 2

gregkh, 2016-12-07:
Sounds like ubuntu is messed up :)

If you have patches that resolve these things, please submit them, but given that I don't use that distro, I can't do anything on my own, sorry.

mailinglists35, 2016-12-07:
my goal was just to get a working lsusb.py - found my way out by just running make lsusb.py :D

gregkh, 2016-12-07:
Ok, closing out as your problem is resolved :)

closed 2016-12-07 by gregkh

mailinglists35, 2016-12-07:
uhm, maybe should be kept open, perhaps some ubuntu dev finds this issue. i see they have packaged version 007 and there is a request to package 008 https://bugs.launchpad.net/ubuntu/+source/usbutils/+bug/1480759

gregkh, 2016-12-07:
008 is two years old, things will be different for 008 and for the next release.

gregkh, 2016-12-07:
i should do a new release soon to let distros sync up...