gregkh/usbutils #56: feature request: make lsusb.py display current and default kernel driver for each device

mailinglists35, 2017-09-27:
a very detailed explanation how can the current driver in use be determined and also what is the default driver that the kernel would use for a particular usb device:
https://unix.stackexchange.com/questions/60078/find-out-which-modules-are-associated-with-a-usb-device

would be awesome to see this in the lsusb.py output, as well as the pid of the process if the driver is usbfs.

gregkh, 2017-09-27:
'lsusb -t' already shows you this, as does 'lsusb.py -i'

So I think there's nothing to do here. :)

gregkh, 2017-09-27:
Note, it does not show the userspace pid, sorry, that's not implemented, but if someone wants to add it, I am always willing to review patches.

closed 2017-09-27 by gregkh

mailinglists35, 2017-09-27:
Thank you, I must have not noticed the `-i` parameter, as the description does not lead me to believe it's actually showing the kernel module (it says " -i display interface information"); perhaps could that be improved to say "display kernel module driver"?

`-i` however does only show the current kernel driver; the steps detailed in the stackexchange answer also show how to see what the kernel would use by default, and also shows the pid for usbfs.

gregkh, 2017-09-27:
> Thank you, I must have not noticed the -i parameter, as the description does
> not lead me to believe it's actually showing the kernel module (it says " -i
> display interface information"); perhaps could that be improved to say "display
> kernel module driver"?

Drivers bind to USB interfaces, not devices.  It's not obvious until you
notice your physical device has 5 different drivers bound to it...

> -i however does only show the current kernel driver; the steps detailed in the
> stackexchange answer also show how to see what the kernel would use by default,
> and also shows the pid for usbfs.

The pid is interesting, again, I'll take patches if someone makes
them.

Digging through the modinfo output isn't worth it for lsusb, other tools
can do that better, and it really isn't all that useful in the end.