gregkh/usbutils #73: Make usbdevices use /bin/sh

rossburton, 2018-07-27:
From a glance usbdevices doens't *need* bash, so can it use the #! /bin/sh instead of bash?

gregkh, 2018-07-27:
Why does it matter?  What systems run this that do not have bash but have /bin/sh only?

rossburton, 2018-07-27:
On 27 July 2018 at 13:05, Greg Kroah-Hartman <notifications@github.com>
wrote:

> Why does it matter? What systems run this that do not have bash but have
> /bin/sh only?
>

Embedded systems which are running fine without bash and adding this
package for debugging doesn't need to pull in bash.  Counter-argument: why
not?  It's not got bashisms, so why not use plain sh?

Ross

rossburton, 2018-07-27:
```
$ bash ./usb-devices >with-bash
$ dash  ./usb-devices >with-dash
$ busybox sh ./usb-devices >with-busybox
$ cat with-bash

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480 MxCh= 4
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev=04.09
S:  Manufacturer=Linux 4.9.0-6-amd64 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=0000:05:00.0
C:  #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=0mA
I:  If#=0x0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
[snip]

$ md5sum with-*
deaf8027fb84b860bfa607ab98ba5255  with-bash
deaf8027fb84b860bfa607ab98ba5255  with-busybox
deaf8027fb84b860bfa607ab98ba5255  with-dash
```

PR #74 (2018-07-27): usb-devices: use /bin/sh hashbang
This script doesn't use any bashisms, so change the hashbang to /bin/sh.

Fixes #73.
commit 508d1acf (2018-07-27): usb-devices: use /bin/sh hashbang

This script doesn't use any bashisms, so change the hashbang to /bin/sh.

Fixes #73.
closed 2018-07-27 by gregkh