gregkh/usbutils #90: iconv / libiconv detection problem (buildroot, uClibc)

Nazdravi, 2019-04-11:
When compiling usbutils under buildroot I got linker errors because of missing "libiconf_open" etc.. This patch solved the issue in my case (adapted from libglib2):
```
--- usbutils-010.original/configure.ac	2018-05-15 16:15:34.000000000 +0200
+++ usbutils-010/configure.ac	2019-04-11 14:54:18.841489702 +0200
@@ -13,7 +13,20 @@
 AC_SYS_LARGEFILE
 
 AC_CHECK_HEADERS([byteswap.h])
-AC_CHECK_FUNCS([nl_langinfo iconv])
+
+found_iconv=no
+# Check if we have GNU libiconv
+if test $found_iconv = "no"; then
+        AC_CHECK_LIB(iconv, libiconv_open, [with_libiconv=gnu; found_iconv=yes])
+fi
+# Check in the C library first
+if test $found_iconv = "no"; then
+        AC_CHECK_FUNC(iconv_open, [with_libiconv=no; found_iconv=yes])
+fi
+# Check if we have a iconv in -liconv, possibly from vendor
+if test $found_iconv = "no"; then
+        AC_CHECK_LIB(iconv, iconv_open, [with_libiconv=native; found_iconv=yes])
+fi
 
 PKG_CHECK_MODULES(LIBUSB, libusb-1.0 >= 1.0.9)
 
```

gregkh, 2019-04-16:
Care to make a real patch and submit it either through github, or by email, so we can resolve this?

gregkh, 2019-04-16:
And this also is odd, buildroot has been building usbutils for years without this being needed, what changed?

gregkh, 2019-05-06:
Closed due to lack of response.  Will be glad to help out, just answer the above and reopen this.

closed 2019-05-06 by gregkh