gregkh/usbutils #92: USB device(s) not listing correctly

martin-minovski, 2019-04-22:
We're struggling to get a fingerprint scanner working on Ubuntu (also Debian etc.); and as one person suggested, the issue might be related to usbutils not listing the device properly - https://bugs.launchpad.net/usbutils/+bug/1721909/
I am not an expert myself in linux driver development, yet there are quite a lot of people looking for a fix. lsusb shows either one of these lines:

Bus 001 Device 005: ID 27c6:5381
or Bus 001 Device 003: ID 27c6:5395
or Bus 001 Device 004: ID 27c6:5301
or Bus 001 Device 003: ID 27c6:5395
or Bus 001 Device 004: ID 27c6:538c
or Bus 001 Device 007: ID 27c6:5740
etc.

Also, as another person pointed out, the vendor id 27c6 should at least say "Shenzhen Huiding" instead of an empty string, as per https://github.com/torvalds/linux/blob/e19b205be43d11bff638cad4487008c48d21c103/Documentation/devicetree/bindings/vendor-prefixes.txt

Any help will be long appreciated.

ValZapod, 2019-04-22:
It should not say your vendor, because lsusb is using usb.ids. Update usb.ids through 
`sudo update-usbids`

ValZapod, 2019-04-22:
Also please keep in mind that you can delete `apt remove usbutils`. It is not part of kernel and just an utility.

gregkh, 2019-05-06:
As was stated, usbutils has nothing to do with your device working properly or not.  Fingerprint scanners use the userspace fingerprint scanner library and utility, which is independent of usbutils.

The vendor id string comes directly from the device itself, so if it reports it incorrectly, you should fix the device.  And, you can update the usb.ids file, as noted above as to how to do.

Closing this out now, as it doesn't pertain to this program, thanks.

closed 2019-05-06 by gregkh