gregkh/usbutils #95: usbutils-011 release tarball incomplete

Polynomial-C, 2019-05-07:
I found two issues with `011` release tarball:

- there is no pre-generated `configure` file
- the `usbhid-dump` folder is empty which requires net access (`git submodule update ...`) in order to successfully bootstrap the package. I do not expect that from a release tarball.

gregkh, 2019-05-07:
Yes, I know.  We are working on switching to the "clean" way of doing releases on kernel.org, and that means the tarball is generated directly from the git tree, not relying on me to do a sync and create some random tarball on my laptop.

the pre-generated configure file should be fine to fix, just run 'autogen.sh'

the usbhid-dump directory being empty, that's a bigger issue.  Look at the new branch here for a potential way to resolve this, by moving the usbhid-dump repo into this one directly.

Hopefully this can be resolved soon, please give us some time...

gregkh, 2019-05-07:
Ok, usbhid-dump is now part of the main repo, and is in the 012 release.

So all should be good here.  See the release announcement for all of the details here, and how not having 'configure' is actually a good thing.

closed 2019-05-07 by gregkh