gregkh/usbutils #102: lsusb for windows

Roman-Koshelev, 2019-12-10:
Why not lsusb under windows? You can collect it with minimal edits ...

gregkh, 2019-12-10:
I can collect what with minimal edits?

This is a tool for Linux and libusb-based systems.  If you have patches to get it to work on Windows, by all means submit them here, otherwise this is not a real "issue" for me.

closed 2019-12-10 by gregkh

Roman-Koshelev, 2019-12-12:
libusb works on windows too. I ran lsusb under windows and linux. You can see the difference yourself (there is almost none). [https://yadi.sk/d/ouWXtDHKgwlBwA](url)

ValZapod, 2020-07-02:
But you will need an instruction how to compile with Visual Studio 2019, also please instead of Yandex, attach it here on github (drag and drop zif file)! This is cool, like I did to edid-decode library.