gregkh/usbutils #111: lsbusb incorrectly reports "descriptor too short" for video PROCESSING unit with bsdUVC 0x0100

massweet, 2020-08-08:
Looking in the kernel driver source, bmVideoStandards was added to 0x0110, and so if bsdUVC is less than this the processing unit descriptor should be a minimum of 9 octets, and not include this field.
See attached unified diff for a possible fix.
[lsbusb.c.diff.txt](https://github.com/gregkh/usbutils/files/5045163/lsbusb.c.diff.txt)

massweet, 2020-08-09:
Oh - feel free to use the patch, under your current GPL 3.0 licensing.

gregkh, 2020-10-22:
The license for this project is GPLv2 only, sorry, if you submit this for GPLv3, I can't take it.

Please submit as a pull request, or submit the patch in email and I will be glad to review it that way.

closed 2020-10-22 by gregkh

ValZapod, 2021-11-02:
I suppose the user just did not know that Linux kernel is not GPL2.0+ but only GPL2.0. After all this is not in a license but in source files. @massweet You need to clarify that you agree with GPL 2.0 only not "GPL 2.0 and later".