gregkh/usbutils #115: Not working with raspberry pi Zero w ?

geminigeek, 2020-10-14:
hi,

i tried to use usbreset  on a huawei usb 4g modem after using usb_modswitch its not working ?

the command to usb_modswitch is below 
usb_modeswitch -v 12d1 -p 157c -P 1506 -M 55534243123456780000000000000011062000000100000000000000000000 -i 2

here is output of the process i tried 

root@PIZeroBot:~/reset2# lsusb
Bus 001 Device 004: ID 12d1:1506 Huawei Technologies Co., Ltd. Modem/Networkcard
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@PIZeroBot:~/reset2# ./usbreset 12d1:1506
Resetting HUAWEI_MOBILE ... ok
root@PIZeroBot:~/reset2# lsusb
Bus 001 Device 004: ID 12d1:1506 Huawei Technologies Co., Ltd. Modem/Networkcard
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Regards
Dee

gregkh, 2020-10-14:
On Wed, Oct 14, 2020 at 06:17:00AM -0700, geminigeek wrote:
> hi,
> 
> i tried to use usbreset  on a huawei usb 4g modem after using usb_modswitch its not working ?

It is never guaranteed to work at all, you need a USB hub that supports
this feature.  I do not see a hub in your USB chain (root hubs do not
count).

thanks,

greg k-h

geminigeek, 2020-10-14:
hi,

tried with a hub

root@PIZeroBot:~# lsusb
Bus 001 Device 007: ID 12d1:1506 Huawei Technologies Co., Ltd. Modem/Networkcard
Bus 001 Device 005: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@PIZeroBot:~#  ./reset2/usbreset 001/007
Resetting HUAWEI_MOBILE ... ok
root@PIZeroBot:~# lsusb
Bus 001 Device 007: ID 12d1:1506 Huawei Technologies Co., Ltd. Modem/Networkcard
Bus 001 Device 005: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

geminigeek, 2020-10-14:
hi

Sorry for the confusion if any

let me suggest what i want to achieve . when i insert the usb modem it shows up as usb pendrive. i usb_modeswitch it to modem mode . with usbreset i was hopping the modem will come back to pendrive mode as it happens when i physically plug out and plug in the modem .

there is progress seems the modem is getting a device id now

root@PIZeroBot:~/reset2# lsusb
Bus 001 Device 009: ID 12d1:1506 Huawei Technologies Co., Ltd. Modem/Networkcard
Bus 001 Device 004: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@PIZeroBot:~/reset2# ./usbreset 001/004
Resetting USB 2.0 Hub [MTT] ... ok
root@PIZeroBot:~/reset2# lsusb
Bus 001 Device 010: ID 12d1:1506 Huawei Technologies Co., Ltd. Modem/Networkcard
Bus 001 Device 004: ID 1a40:0201 Terminus Technology Inc. FE 2.1 7-port Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

gregkh, 2020-10-14:
On Wed, Oct 14, 2020 at 07:51:13AM -0700, geminigeek wrote:
> hi
>
> Sorry for the confusion if any
>
> let me suggest what i want to achieve . when i insert the usb modem it
> shows up as usb pendrive. i usb_modeswitch it to modem mode . with
> usbreset i was hopping [...]

I have no idea if that would work, you really are on your own here,
usbreset is a "you are lucky if it works" type of program, because it is
not required to be supported by USB hardware.

good luck!

greg k-h

gregkh, 2020-10-22:
Closing out here, usbreset is a "user beware" program, as it says.

closed 2020-10-22 by gregkh