gregkh/usbutils #123: syntax error in autogen.sh

Duhjoker, 2021-04-13:
hi 

  getting a syntax error when using autogen.sh for configure but cant figure out how to fix 

./configure: line 4500: syntax error near unexpected token `LIBUSB,'
./configure: line 4500: `PKG_CHECK_MODULES(LIBUSB, libusb-1.0 >= 1.0.9)'

gregkh, 2021-04-13:
Do you have libusb installed properly?

What distro is this?  What version of libusb?

Duhjoker, 2021-04-18:
Yes I'm using a proot of Ubuntu 20.04. I have used the instruction here https://zoomadmin.com/HowToInstall/UbuntuPackage/libusb-1.0-0

gregkh, 2021-04-19:
You probably also need the libusb-dev package if you are going to build something against it.

Why not just use the usbutils version that comes with the distro?

tormodvolden, 2021-11-19:
If only libusb-dev was missing there would be a proper error message from configure, and not a syntax error. This is probably due to missing installation of pkg-config (which includes pkg.m4). Anyway it also looks like a stale tree: libusb 1.0.9 was the requirement only up to version 0.10, back in 2018. FYI Ubuntu 20.04 ships with version 012.

gregkh, 2023-10-12:
Ok, not having `pkg-config installed isn't a usbutils issue, now closing, thanks!

closed 2023-10-12 by gregkh