gregkh/usbutils #132: When is next release/tag planned?

grundlerchromium, 2021-07-29:
vapier@chromium.org and I are wondering if there is a plan for the next release.

It appears the most recent release was Oct 2020 (V013) and the repo has nearly 30 changes since then:
grundler <2241>git log --oneline v013.. | fgrep -v "Merge pull request" | wc -l
29

thanks,
grant

gregkh, 2021-07-29:
I can do a release soon.  Let me make sure all pending work is caught up on, normally I do one release a year or so for this slow-moving codebase :)

grundlerchromium, 2021-07-29:
Ok - thanks - no stress to release though. Not looking for a regular candence - just understand roughly when you feel ready to "ship" the next release.

gregkh, 2021-07-29:
Usually when someone asks me because a feature was added that they need.  No regular cadence at all here.

gregkh, 2022-03-12:
A release happened since then, closing this out.

closed 2022-03-12 by gregkh