gregkh/usbutils #140: README lacks statement to origin of code

hadmut, 2022-02-26:
Hi,

it's easy to get the source code for usbutils, e.g. under debian/ubuntu just do an apt source usbutils. 

But then, the code itself does not contain any hint about it's origin, where to get the original source, where to open issues, i.e. a link to this github repository here. It's just the debian patch which tells in debian/control, where the source came from, or the allmighty Google. 

Usually source codes contain some statement, usually in the README, about where their home is. 

regards

gregkh, 2022-05-18:
Home can be wherever you find the code :)

I'll update this and add a link, thanks.

gregkh, 2023-10-12:
Now resolved in commit db068dbef5935242bba6a2f9fef657c0a2e663dc