gregkh/usbutils #142: USB device often named incorrectly

meetnick, 2022-03-11:
I am facing some weird issue on the output from **lsusb**, appending **HD USB Camera** to some USB devices names, as can be seen below:
```
Bus 001 Device 011: ID 32e4:9520 HD Camera Manufacturer HD USB Camera
Bus 001 Device 014: ID 060e:0960 Transmonde Technologies, Inc. HD USB Camera
Bus 001 Device 019: ID 0b05:1939 ASUSTek Computer, Inc. HD USB Camera
Bus 001 Device 018: ID 04b8:0e28 Seiko Epson Corp. HD USB Camera
```
this is from usbutils 012, as below:
```
lsusb --version
lsusb (usbutils) 012
```

Is there any work around this to have USB devices properly named?

gregkh, 2022-03-12:
Version 012 was from 2019.  This issue should be fixed in the latest version, please upgrade and see if it resolves the issue or not for you.

meetnick, 2022-03-12:
It really fixed it after upgrading to latest 014. Do you have the commit hash which fixed it for reference?

Thank you so much, by the way

gregkh, 2022-03-12:
I do not remember, sorry.  Look in the git history, I think it was a number of different commits to do so.  Just update to the latest version, that's the better thing to always do anyway, never cherry-pick things.

closed 2022-03-12 by gregkh