gregkh/usbutils #143: Why the hub port number is hardcoded as 1?

smrtos, 2022-03-20:
This lsusb always treat the port number of hub as 1.
While the lsusb.py treat the port number of hub as 0.

Related code:
https://github.com/gregkh/usbutils/blob/4bf4819a0dfddc53810a640e3be83dbbeeab0802/lsusb-t.c#L163

Related output of lsusb:
[image]

Related output of lsusb.py:
[image]

I think they should be consistent at least...

gregkh, 2022-03-20:
lsusb.py is showing you the interface for that device, you asked for it with the "-I" option.  If you don't want to see it, don't add that option to the command line when you run it.

As far as I can tell, this is working as designed.  If not, please provide the version of the release you are using, as well as the full command line arguments you are using to see the above two images.

thanks!

closed 2022-03-20 by gregkh

smrtos, 2022-03-21:
I am using the `lsusb (usbutils) 014`. It should be the latest one as of now.
I don't see the version of `lsusb.py` but it should be the same.

I am using `lsusb.py -I`.

According to [here](https://www.kernel.org/doc/Documentation/ABI/stable/sysfs-bus-usb), the general format is:

`<bus>-<port[.port[.port]]>:<config>.<interface>`

So in the lsusb.py output, the `3-0:1.0` should mean `bus 3, port 0, config 1, interface 0`.

So, 
lsusb shows port 1, 2, 10 for usb3.
lsusb.py shows port 0, 2, 10 for usb3.

I think there must be one showing the wrong port.

gregkh, 2022-03-21:
The sysfs documentation is for the values that the kernel exports in sysfs, and has nothing to do with the lsusb.py output format.

You are showing the interfaces in lsusb.py, and the values are shown there correctly from what I can determine.

I do not see the discrepancy here, what do you expect the output should look like?

smrtos, 2022-03-21:
I am expecting the same set of port numbers to be shown for each USB hub. But maybe I am wrong since I am not an USB export.

gregkh, 2022-03-21:
Again, the lsusb.py tool is showing you the interface number here, not the port number at all.

The sysfs documentation you are pointing at is how sysfs exports the values, NOT how the lsusb.py tool is showing the values.  They are independent and not related as you have pointed out.

smrtos, 2022-03-21:
Thanks. It seems  different schemas are used.

Could you give a formal definition of the lsusb.py output like `3-2:1.3`?

If the component `2` in the `3-2:1.3` means interface, what does the last `3` mean?

gregkh, 2022-03-21:
I think it is the endpoint number.  Would have to go dig through the python source to verify.

smrtos, 2022-03-21:
I guess I need to dig into some USB book as well.

smrtos, 2022-09-08:
> I think it is the endpoint number. Would have to go dig through the python source to verify.

Hi, could you provide the formal definition of the `3-2:1.3` now?