gregkh/usbutils #144: UAC2: lsusb -v only show first string descriptor when there are several ones

nodens, 2022-03-31:
Hi,

trying to implement an alsa UCM2 profile for my USB audio interface (Arturia Minifuse 2), I noticed that when using lsusb -v, I can only see the first channel description in iChannelNames attribute from a cluster.

e.g:
```
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       2
      bNumEndpoints           2
      bInterfaceClass         1 Audio
      bInterfaceSubClass      2 Streaming
      bInterfaceProtocol     32 
      iInterface              4 MiniFuse 2 
      AudioStreaming Interface Descriptor:
        bLength                16
        bDescriptorType        36
        bDescriptorSubtype      1 (AS_GENERAL)
        bTerminalLink           2
        bmControls           0x00
        bFormatType             1
        bmFormats          0x00000001
          PCM
        bNrChannels             4
        bmChannelConfig    0x00000000
        iChannelNames          13 Output Left

```

Dumping the usb data under windows, I see all the iChannelNames, there is one per channel (so 4 here). In this case, it should show
`Output Left, Output Right, Loopback Left, Loopback Right` or something similar.

I suppose the issue is only the first string descriptor (array[0]) is shown, where it should iterate on bNrChannels to get array[1-3] descriptors as well.

Sadly, I'm not fluent enough in C to see where in the code it could be fixed ;)

Thanks!

nodens, 2022-03-31:
digging into it a bit, it seems channel names for UAC are hardcoded in `desc-defs.c` ?

nodens, 2022-05-23:
Hi!

I use usbutils / lsusb 014 (from Debian unstable).

The problem is I see only one descriptor per output (first), instead of all the strings per channel, which would be useful to handle alsa configuration (use-case manager, etc).
So here I see Output Left, which is one channel, but I'd like to know the 3 other channels are Output Right, Loopback Left, Loopback Right.

To get all the channel descriptors, I had to run a windows VM and dump the data there:
[Win10_DescriptorDump_MiniFuse_2.txt](https://github.com/gregkh/usbutils/files/8752764/Win10_DescriptorDump_MiniFuse_2.txt)
.
complete lsusb -v for the same device (as root): [lsusb_minifuse2.txt](https://github.com/gregkh/usbutils/files/8752806/lsusb_minifuse2.txt)

The windows dump shows the index of descriptor. With another tool (usblyzer, but it was a trial version and I can't export it anymore now), I saw it neatly organized per AudioControl descriptor.

ValZapod, 2022-06-03:
> Dumping the usb data under windows

Are you using USB Device Tree Viewer, which is actually on par with usbutils on windows?

ValZapod, 2022-06-03:
@gregkh bmChannelConfig is not the same as iChannelNames.

bmChannelConfig is standard channels, this is lile standard 24 channels of 22.4 or 8 channels of 7.1 or 6 channels of 5.1.

This is like
          Front Left (FL)
          Front Right (FR)
          Front Center (FC)
          Low Frequency Effects (LFE)
          Back Left (BL)
          Back Right (BR)
          Front Left of Center (FLC)
          Front Right of Center (FRC)

so 7.1. But, there is also a mic possible and some loops. Those are done with iChannelNames.

Other channels are possible https://en.wikipedia.org/wiki/22.2_surround_sound (so as you can see in Adobe Audiotion TpSiR, TpBC, BtFL, etc, etc).

And some other like Auro 3D object audio can describe an object for perfect over the head loudspeaker which is called the **Voice Of God**. This is also non standard, obviosuly.

nodens, 2022-06-05:
> > Dumping the usb data under windows
> 
> Are you using USB Device Tree Viewer, which is actually on par with usbutils on windows?

Well, now I am, thanks for the hint!
Here is the dump.
[Win10_usbdtreeviewDump_Minifuse_2.txt](https://github.com/gregkh/usbutils/files/8840486/Win10_usbdtreeviewDump_Minifuse_2.txt)

closed 2024-09-02 by gregkh