gregkh/usbutils #146: lsusb names are mixed

Lowrida, 2022-05-18:
I had a video capture device without a name and a problem of mixed up names of my keyboard with webcam and phone..
 
[image]

So i installed **20-usb-vendor-model.hwdb** to **/etc/udev/hwdb.d** and updated the database. Now my video capture device has a name, but the mixing of names from my keyboard (green) and the creative webcam (red) is still an issue. The mixing of names with keyboard an my Huawei P10 is gone now!
 
[image]
 
Any idea what this could be? Something wrong on my system or a bug??
I am on Linux Mint 20.3 with kernel 5.14 and lsusb version 012.

gregkh, 2022-05-18:
lsusb 012 is from 2019, this issue should already be fixed, please update to 014 and see if it is still there or not.

If it is an issue with 014, please reopen, thanks!

closed 2022-05-18 by gregkh

Lowrida, 2022-05-18:
014 solved the problem! Thanks.