gregkh/usbutils #150: heap-buffer-overflow when getting verbose info of a usb camera device

qiankehan, 2022-06-20:
The camera device is:
```
Bus 001 Device 004: ID 13d3:56bb IMC Networks Integrated Camera
```
Version:
usbutils v014-7-g77b93c9  
libusb1-1.0.25-8.fc36.x86_64

Steps:
1. Compile the code with address sanitizer: `CFLAGS=' -fsanitize=address -fno-omit-frame-pointer -static-libasan -g' ./configure && make`
2. Get the verbose info of **13d3:56bb**: `ASAN_OPTIONS=fast_unwind_on_malloc=0 ./lsusb -d 13d3:56bb -v`. Then get the heap-buffer-overflow report:
```

Bus 001 Device 004: ID 13d3:56bb IMC Networks Integrated Camera
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x13d3 IMC Networks
  idProduct          0x56bb
  bcdDevice           19.02
  iManufacturer           2 Azurewave
  iProduct                1 Integrated Camera
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0334
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0
      iFunction               5 Integrated Camera
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0
      iInterface              5 Integrated Camera
      VideoControl Interface Descriptor:
        bLength                13
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.00
        wTotalLength       0x004e
        dwClockFrequency       15.000000MHz
        bInCollection           1
        baInterfaceNr( 0)       1
      VideoControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      3 (OUTPUT_TERMINAL)
        bTerminalID             4
        wTerminalType      0x0101 USB Streaming
        bAssocTerminal          0
        bSourceID               3
        iTerminal               0
      VideoControl Interface Descriptor:
        bLength                27
        bDescriptorType        36
        bDescriptorSubtype      6 (EXTENSION_UNIT)
        bUnitID                 3
        guidExtensionCode         {28f03370-6311-4a2e-ba2c-6890eb334016}
        bNumControls           16
        bNrInPins               1
        baSourceID( 0)          2
        bControlSize            2
        bmControls( 0)       0x9f
        bmControls( 1)       0x1e
        iExtension              0
      VideoControl Interface Descriptor:
        bLength                18
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0201 Camera Sensor
        bAssocTerminal          0
        iTerminal               0
        wObjectiveFocalLengthMin      0
        wObjectiveFocalLengthMax      0
        wOcularFocalLength            0
        bControlSize                  3
        bmControls           0x0020000e
          Auto-Exposure Mode
          Auto-Exposure Priority
          Exposure Time (Absolute)
      VideoControl Interface Descriptor:
        bLength                11
        bDescriptorType        36
        bDescriptorSubtype      5 (PROCESSING_UNIT)
      Warning: Descriptor too short
        bUnitID                 2
        bSourceID               1
        wMaxMultiplier          0
        bControlSize            2
        bmControls     0x0000157f
          Brightness
          Contrast
          Hue
          Saturation
          Sharpness
          Gamma
          White Balance Temperature
          Backlight Compensation
          Power Line Frequency
          White Balance Temperature, Auto
=================================================================
==1351035==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x60700000076e at pc 0x000000418d25 bp 0x7ffc63a6f010 sp 0x7ffc63a6f008
READ of size 1 at 0x60700000076e thread T0
    #0 0x418d24 in dump_videocontrol_interface (/home/hhan/Software/usbutils/lsusb+0x418d24)
    #1 0x409ade in dump_altsetting (/home/hhan/Software/usbutils/lsusb+0x409ade)
    #2 0x40a527 in dump_interface (/home/hhan/Software/usbutils/lsusb+0x40a527)
    #3 0x408840 in dump_config (/home/hhan/Software/usbutils/lsusb+0x408840)
    #4 0x432725 in dumpdev (/home/hhan/Software/usbutils/lsusb+0x432725)
    #5 0x4334c8 in list_devices (/home/hhan/Software/usbutils/lsusb+0x4334c8)
    #6 0x433ed4 in main (/home/hhan/Software/usbutils/lsusb+0x433ed4)
    #7 0x7ff6efafe54f in __libc_start_call_main (/lib64/libc.so.6+0x2954f)
    #8 0x7ff6efafe608 in __libc_start_main_alias_1 (/lib64/libc.so.6+0x29608)
    #9 0x403664 in _start (/home/hhan/Software/usbutils/lsusb+0x403664)

0x60700000076e is located 0 bytes to the right of 78-byte region [0x607000000720,0x60700000076e)
allocated by thread T0 here:
    #0 0x7ff6f040e68f in __interceptor_malloc (/lib64/libasan.so.8+0xba68f)
    #1 0x7ff6f033b16c in raw_desc_to_config (/lib64/libusb-1.0.so.0+0x616c)

SUMMARY: AddressSanitizer: heap-buffer-overflow (/home/hhan/Software/usbutils/lsusb+0x418d24) in dump_videocontrol_interface
Shadow bytes around the buggy address:
  0x0c0e7fff8090: fd fd fd fd fd fd fd fd fd fd fa fa fa fa fd fd
  0x0c0e7fff80a0: fd fd fd fd fd fd fd fd fa fa fa fa fd fd fd fd
  0x0c0e7fff80b0: fd fd fd fd fd fd fa fa fa fa 00 00 00 00 00 00
  0x0c0e7fff80c0: 00 00 00 00 fa fa fa fa fd fd fd fd fd fd fd fd
  0x0c0e7fff80d0: fd fd fa fa fa fa fd fd fd fd fd fd fd fd fd fd
=>0x0c0e7fff80e0: fa fa fa fa 00 00 00 00 00 00 00 00 00[06]fa fa
  0x0c0e7fff80f0: fa fa fd fd fd fd fd fd fd fd fd fd fa fa fa fa
  0x0c0e7fff8100: 00 00 00 00 00 00 00 00 00 03 fa fa fa fa fa fa
  0x0c0e7fff8110: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c0e7fff8120: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c0e7fff8130: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==1351035==ABORTING
```

gregkh, 2023-10-12:
Your descriptor is invalid, the error message right before the splat gives you a hint:
`      Warning: Descriptor too short`

Is this a shipping device?  We warn that the descriptor is invalid, and do our best to print stuff out, but then we run off the end of the buffer of the descriptor trying to print out information that should be there.  I don't know how "robust" we really need to make the tool here as this doesn't really cause any problems, right?

qiankehan, 2023-10-16:
> Your descriptor is invalid, the error message right before the splat gives you a hint: ` Warning: Descriptor too short`
>
> Is this a shipping device? We warn that the descriptor is invalid, and do our [...]

No. It is an integrated laptop camera. This issue didn't cause any problems. Since I have retired that laptop, I cannot reproduced it now.

gregkh, 2023-10-16:
Thanks for letting me know, closing this out now

closed 2023-10-16 by gregkh