gregkh/usbutils #152: lsusb outputs not match the usbutils version

lunarwhite, 2022-08-11:
Hi, I try to install and explore the usage of usbutils in two different ways:

## 1 install from apt

### install
```shell
$ sudo apt update
$ sudo apt install usbutils -y
```

### use
```shell
$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

$ lsusb -V
lsusb (usbutils) 007

``` 

## 2 make and install from source code

### install & make
```shell
$ sudo apt update
$ sudo apt install git gcc make autoconf pkg-config libusb-1.0-0-dev libudev-dev -y
$ git clone https://github.com/gregkh/usbutils.git && cd usbutils/ 
$ git submodule init && git submodule update
$ autoreconf --install --symlink && ./configure
$ make && make install

```

### use
```shell
$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux 5.13.0-51-generic ehci_hcd EHCI Host Controller
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. VMware Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware VMware Virtual USB Mouse
Bus 002 Device 001: ID 1d6b:0001 Linux 5.13.0-51-generic uhci_hcd UHCI Host Controller

$ lsusb -V
lsusb (usbutils) 014
```

---

My env:
`Linux 3f0a1abbf23a 5.13.0-51-generic #58~20.04.1-Ubuntu SMP x86_64 x86_64 x86_64 GNU/Linux`

And I use docker for these tests:
`docker run -it ubuntu:18.04`

My questions:

1. In way-1, why I already `apt update`, but usbutils installed version is `007` which is outdated?
2. In way-2, though the version `014` is up-to-date, but the `lsusb` outputs are not up-to-date? I go to check the usb.ids in http://www.linux-usb.org/usb.ids, and I find the outputs don't match the latest usb.ids.
3. How can I get the way-1's `lsusb` outputs through way-2's installation steps?

Is there something I misunderstood? 

Thanks in advance :)

gregkh, 2022-08-11:
version 007 is from 2013, please work with your distro to use a more up to date version, nothing I can do about that.

And yes, newer versions of lsusb use better device id information, it relies on the hw.ids database. 

Please work with your distro to resolve this, it is not an upstream issue.  Use the support contract you are paying them for to help resolve this issue.  Good luck!

closed 2022-08-11 by gregkh

lunarwhite, 2022-08-11:
Thanks for your reply. I've fixed the `007` version problem by using latest Ubuntu verison. But the `lsusb` outputs are still outdated.

So I want to figure out, if I want to get the latest device id information(like way-1 outputs), what depenencies must be installed? Did I miss any package? 

Or the output just rely on the hw.ids database, not the usbutils? Should I try to update the database manually？

gregkh, 2022-08-11:
I do not know, please work with your distribution on this, it all depends on how they have packaged this tool as to where it finds the information from.

lunarwhite, 2022-08-15:
[update] issue solved, share the solution for someone who got same error.

`usbutils` has these dependencies:
- `usb.ids`
- `hwdb.bin`

Make sure you have these installed. For me, `apt install udev` will install the `hwdb.bin`.

Find more details in https://unix.stackexchange.com/questions/621265/how-to-use-lsusb-in-a-docker-container-to-get-usb-device-descriptions