gregkh/usbutils #153: Offered usbutils.pc.in file for inclusion... or not

DaemonDave, 2022-08-19:
```
# An added .pc.in file  :  by DRE 2022  (https://github.com/DaemonDave)

# The Folder Variables 
prefix=@prefix@
exec_prefix=@prefix@
libdir=@prefix@/lib64
includedir=@prefix@/include

# Package Name that may change but requires no update
Name: @PACKAGE_NAME@					 		# human-readable name
Description: usbutils    This is a collection of USB tools for use on Linux and BSD systems\
 to query what type of USB devices are connected to the system. This is to be run on a USB host\
  (i.e. a machine you plug USB devices into), not on a USB device (i.e. a device you plug into a USB host.) # human-readable description
# Version updated everytime
Version: @VERSION@
# where to find the repository?
URL: https://github.com/gregkh/usbutils
# Requires: is from configure.ac
  Requires: libusb-1.0 >= 1.0.14  libudev >= 196
  Conflicts:
  Libs: -L@prefix@/lib64
  Libs.private: @LIBUSB_LIBS@ @UDEV_LIBS@
  Cflags: @CFLAGS@ @LIBUSB_CFLAGS@ @UDEV_CFLAGS@
```
  github doesn't allow .pc.in files so above is the file contents.  I placed it in the top folder.  

If you want to make this pkgconfig file then replace these lines into your configure.ac and you are finished!
```

AC_CONFIG_FILES([
    usbutils.pc
    Makefile
])
```

Above is a proposed usbutils.pc.in package-config configuration file to be included. I made it because the udev copy I am using requires configure.ac looks for a usbutils.pc file for some creepy hidden variable looking....

Adopt or ignore at your leisure. 

Cheers!

Daemondave

gregkh, 2022-08-25:
Can you turn this into a normal patch and submit it that way so I can easily accept it?  Otherwise trying to cut/paste from a github issue is almost impossible...

gregkh, 2023-10-12:
Now merged.

closed 2023-10-12 by gregkh