gregkh/usbutils #154: USB MIDI 2.0 Descriptors

symdeb, 2022-08-25:
This is output from lsusb on Ubuntu 22.04.01. This is a prototype MIDI 2.0  (UMP)  USB device using alternate setting 1 descriptors
Descriptor type 37 subtype 2 is supposed to list bNumGrpTrmBlock but it shows  MIDIJack.
Reference: USB Device Class Definition for MIDI Devices, Version 2.0 p.42.
Not sure where to post this. Only could find this github repo with libusb sources
Have no reference to other devices as a comparison (no such devices seem to exist yet).

  AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0109
        bInCollection           1
        baInterfaceNr(0)        1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       1
      bNumEndpoints           2
      bInterfaceClass         1 Audio
      bInterfaceSubClass      3 MIDI Streaming
      bInterfaceProtocol      0 
      iInterface              0 
      MIDIStreaming Interface Descriptor:
        bLength                 7
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               2.00
        wTotalLength       0x0007
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
        MIDIStreaming Endpoint Descriptor:
          bLength                 5
          bDescriptorType        37
          bDescriptorSubtype      2 (Invalid)
          bNumEmbMIDIJack         1
          baAssocJackID( 0)       1

symdeb, 2022-08-25:
[MIDI_2_Alt_setting_1.txt](https://github.com/gregkh/usbutils/files/9422500/MIDI_2_Alt_setting_1.txt)

gregkh, 2022-08-25:
This is the correct repo, yes.

I don't understand the problem here, what does the output show and what should it show exactly?

symdeb, 2022-08-25:
This is what the tool reports:

        MIDIStreaming Endpoint Descriptor:
          bLength                 5
          bDescriptorType        37
          bDescriptorSubtype      2 (Invalid)
          bNumEmbMIDIJack         1
          baAssocJackID( 0)       1

[image]

gregkh, 2022-08-25:
Thanks for the info, should now be fixed in the tree in commit f7cb6bcf153d ("lsusb: fix up Midi Device specification devices")

Can you pull the latest tree and build and test it to verify?

symdeb, 2022-08-26:
git checkout f7cb6bc
HEAD is now at f7cb6bc lsusb: fix up Midi Device specification devices
Ther error is gone but the field names are not updated :  bNumEmbMIDIJack ,  baAssocJackID( 0)

symdeb, 2022-08-31:
Posted a pull request for this.

closed 2022-08-31 by symdeb