gregkh/usbutils #161: Bus sort order

cbiedl, 2022-10-11:
See Debian bug report #927188 (<https://bugs.debian.org/927188>): The lsusb seems to list busses in reverse order.
For the simple mode (default), this can be fixed by adding sorter in `list_devices` (patch available upon request).
For the tree mode (-t), I noticed the buses are sorted in reverse order by intention:

```
    static void sort_busses(void)
    {
            /* need to reverse sort bus numbers */
```
[lsusb-t.c:661]

This has been around since the introduction in 2009.
While reversing the sort order is trivial - mind to explain the idea behind this?

ValZapod, 2022-10-27:
Not true, see

Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 004: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 002 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
**Bus 003 Device 002: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub**

gregkh, 2023-10-12:
Should now be resolved, sort order is now low->high, not backwards.  I can't remember why it was originally done that way, sorry.

closed 2023-10-12 by gregkh