gregkh/usbutils #170: The returned value of the “lsusb -h” command is not 0.

Zhanghongtao2417, 2023-06-25:
[root@localhost ~]# lsusb -h
Usage: lsusb [options]...
List USB devices
  -v, --verbose
      Increase verbosity (show descriptors)
  -s [[bus]:][devnum]
      Show only devices with specified device and/or
      bus numbers (in decimal)
  -d vendor:[product]
      Show only devices with the specified vendor and
      product ID numbers (in hexadecimal)
  -D device
      Selects which device lsusb will examine
  -t, --tree
      Dump the physical USB device hierarchy as a tree
  -V, --version
      Show version of program
  -h, --help
      Show usage and help
[root@localhost ~]# echo $?
1
[root@localhost ~]#

gregkh, 2023-06-25:
Ah, never noticed this, let me go fix it up, thanks for the report!

gregkh, 2023-06-25:
Now fixed in commit 882a852d06321968a420fda46aef0c52b8220eae

closed 2023-06-25 by gregkh

Zhanghongtao2417, 2023-06-25:
> Now fixed in commit [882a852](https://github.com/gregkh/usbutils/commit/882a852d06321968a420fda46aef0c52b8220eae)
diff --git a/lsusb.c b/lsusb.c
index 90825c5..f85567e 100644
--- a/lsusb.c
+++ b/lsusb.c
@@ -3797,7 +3797,7 @@ int main(int argc, char *argv[])
                        "  -h, --help\n"
                        "      Show usage and help\n"
                        );
-               return EXIT_FAILURE;
+               return help ? 0 : EXIT_FAILURE;
        }

--

Zhanghongtao2417, 2023-06-25:
> Now fixed in commit [882a852](https://github.com/gregkh/usbutils/commit/882a852d06321968a420fda46aef0c52b8220eae)

diff --git a/lsusb.c b/lsusb.c
index 90825c5..f85567e 100644
--- a/lsusb.c
+++ b/lsusb.c
@@ -3797,7 +3797,7 @@ int main(int argc, char *argv[])
                        "  -h, --help\n"
                        "      Show usage and help\n"
                        );
-               return EXIT_FAILURE;
+               return help ? 0 : EXIT_FAILURE;
        }

Zhanghongtao2417, 2023-06-25:
> Now fixed in commit [882a852](https://github.com/gregkh/usbutils/commit/882a852d06321968a420fda46aef0c52b8220eae)
I'd suggest

 return help ? 0 : EXIT_FAILURE;

gregkh, 2023-06-25:
Oops, right, also we need to check `err`.

And doing `? :` is not ever a good idea, spell out the if statement, makes it easier to maintain over time.

Now fixed in 09af21c154cf