gregkh/usbutils #182: usbutils lsusb devices still referenced upon exit

mcuee, 2023-12-21:
Reference:
* https://github.com/libusb/libusb/issues/1398

As pointed out by @tormodvolden, this is a bug of usbutils and his proposed fix sorted out the issue.

> Isn't this just a bug in usbutils? It uses libusb_get_device_list() so all devices will be reference-counted. In the end it does libusb_free_device_list(list, 0) so without dereferencing the devices. There's no libusb_unref_device() in the code.
> 
> Without looking too close at the code, I guess it becomes correct If this is changed to libusb_free_device_list(list, 1).

mcuee, 2023-12-21:
Run log without the fix.

```
mcuee@UbuntuSwift3 ~/build/libusb/usbutils (master)$ lsusb
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000021] [000044f7] libusb: debug [libusb_init_context] libusb v1.0.27.11855-rc1
[ 0.000056] [000044f7] libusb: debug [usbi_add_event_source] add fd 4 events 1
[ 0.000068] [000044f7] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000072] [000044f7] libusb: debug [usbi_add_event_source] add fd 5 events 1
[ 0.000089] [000044f7] libusb: debug [get_kernel_version] reported kernel version is 5.15.0-91-generic
[ 0.000112] [000044f7] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000116] [000044f7] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000125] [000044f7] libusb: debug [op_init] sysfs is available
[ 0.001178] [000044f8] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.004336] [000044f7] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.004339] [000044f7] libusb: debug [linux_get_device_address] scan usb1
[ 0.004350] [000044f7] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.004351] [000044f7] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.004353] [000044f7] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.004393] [000044f7] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.004395] [000044f7] libusb: debug [linux_get_device_address] scan usb2
[ 0.004402] [000044f7] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.004403] [000044f7] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.004404] [000044f7] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.004441] [000044f7] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
[ 0.004443] [000044f7] libusb: debug [linux_get_device_address] scan 2-1
[ 0.004450] [000044f7] libusb: debug [linux_get_device_address] bus=2 dev=2
[ 0.004451] [000044f7] libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
[ 0.004452] [000044f7] libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
[ 0.004462] [000044f7] libusb: debug [linux_get_parent_info] dev 0x55e6ea4b89c0 (2-1) has parent 0x55e6ea4b8370 (usb2) port 1
[ 0.004492] [000044f7] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.004493] [000044f7] libusb: debug [linux_get_device_address] scan usb3
[ 0.004500] [000044f7] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.004502] [000044f7] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.004504] [000044f7] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.004552] [000044f7] libusb: debug [linux_get_device_address] getting address for device: 3-1 detached: 0
[ 0.004553] [000044f7] libusb: debug [linux_get_device_address] scan 3-1
[ 0.004567] [000044f7] libusb: debug [linux_get_device_address] bus=3 dev=7
[ 0.004568] [000044f7] libusb: debug [linux_enumerate_device] busnum 3 devaddr 7 session_id 775
[ 0.004571] [000044f7] libusb: debug [linux_enumerate_device] allocating new device for 3/7 (session 775)
[ 0.004593] [000044f7] libusb: debug [linux_get_parent_info] dev 0x55e6ea49b000 (3-1) has parent 0x55e6ea4b88c0 (usb3) port 1
[ 0.004630] [000044f7] libusb: debug [linux_get_device_address] getting address for device: 3-1.1 detached: 0
[ 0.004632] [000044f7] libusb: debug [linux_get_device_address] scan 3-1.1
[ 0.004643] [000044f7] libusb: debug [linux_get_device_address] bus=3 dev=8
[ 0.004644] [000044f7] libusb: debug [linux_enumerate_device] busnum 3 devaddr 8 session_id 776
[ 0.004645] [000044f7] libusb: debug [linux_enumerate_device] allocating new device for 3/8 (session 776)
[ 0.004652] [000044f7] libusb: debug [linux_get_parent_info] dev 0x55e6ea49b570 (3-1.1) has parent 0x55e6ea49b000 (3-1) port 1
[ 0.004684] [000044f7] libusb: debug [linux_get_device_address] getting address for device: 3-10 detached: 0
[ 0.004685] [000044f7] libusb: debug [linux_get_device_address] scan 3-10
[ 0.004692] [000044f7] libusb: debug [linux_get_device_address] bus=3 dev=6
[ 0.004693] [000044f7] libusb: debug [linux_enumerate_device] busnum 3 devaddr 6 session_id 774
[ 0.004694] [000044f7] libusb: debug [linux_enumerate_device] allocating new device for 3/6 (session 774)
[ 0.004702] [000044f7] libusb: debug [linux_get_parent_info] dev 0x55e6ea49b220 (3-10) has parent 0x55e6ea4b88c0 (usb3) port 10
[ 0.004732] [000044f7] libusb: debug [linux_get_device_address] getting address for device: 3-2 detached: 0
[ 0.004733] [000044f7] libusb: debug [linux_get_device_address] scan 3-2
[ 0.004740] [000044f7] libusb: debug [linux_get_device_address] bus=3 dev=2
[ 0.004741] [000044f7] libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
[ 0.004742] [000044f7] libusb: debug [linux_enumerate_device] allocating new device for 3/2 (session 770)
[ 0.004749] [000044f7] libusb: debug [linux_get_parent_info] dev 0x55e6ea49bbb0 (3-2) has parent 0x55e6ea4b88c0 (usb3) port 2
[ 0.004780] [000044f7] libusb: debug [linux_get_device_address] getting address for device: 3-3 detached: 0
[ 0.004781] [000044f7] libusb: debug [linux_get_device_address] scan 3-3
[... cut ...]
[ 0.005634] [000044f7] libusb: warning [libusb_exit] device 3.6 still referenced
[ 0.005635] [000044f7] libusb: warning [libusb_exit] device 3.8 still referenced
[ 0.005637] [000044f7] libusb: warning [libusb_exit] device 3.7 still referenced
[ 0.005638] [000044f7] libusb: warning [libusb_exit] device 3.1 still referenced
[ 0.005639] [000044f7] libusb: warning [libusb_exit] device 2.2 still referenced
[ 0.005641] [000044f7] libusb: warning [libusb_exit] device 2.1 still referenced
[ 0.005643] [000044f7] libusb: warning [libusb_exit] device 1.1 still referenced
```

mcuee, 2023-12-21:
Run log with the proposed fix.
```
mcuee@UbuntuSwift3 ~/build/libusb/usbutils (master)$ lsusb
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000021] [00001c02] libusb: debug [libusb_init_context] libusb v1.0.27.11855-rc1
[ 0.000050] [00001c02] libusb: debug [usbi_add_event_source] add fd 4 events 1
[ 0.000061] [00001c02] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.000064] [00001c02] libusb: debug [usbi_add_event_source] add fd 5 events 1
[ 0.000079] [00001c02] libusb: debug [get_kernel_version] reported kernel version is 5.15.0-91-generic
[ 0.000101] [00001c02] libusb: debug [op_init] found usbfs at /dev/bus/usb
[ 0.000104] [00001c02] libusb: debug [op_init] max iso packet length is (likely) 98304 bytes
[ 0.000115] [00001c02] libusb: debug [op_init] sysfs is available
[ 0.000595] [00001c03] libusb: debug [linux_udev_event_thread_main] udev event thread entering
[ 0.004328] [00001c02] libusb: debug [linux_get_device_address] getting address for device: usb1 detached: 0
[ 0.004331] [00001c02] libusb: debug [linux_get_device_address] scan usb1
[ 0.004341] [00001c02] libusb: debug [linux_get_device_address] bus=1 dev=1
[ 0.004342] [00001c02] libusb: debug [linux_enumerate_device] busnum 1 devaddr 1 session_id 257
[ 0.004343] [00001c02] libusb: debug [linux_enumerate_device] allocating new device for 1/1 (session 257)
[ 0.004384] [00001c02] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[ 0.004385] [00001c02] libusb: debug [linux_get_device_address] scan usb2
[ 0.004392] [00001c02] libusb: debug [linux_get_device_address] bus=2 dev=1
[ 0.004393] [00001c02] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[ 0.004394] [00001c02] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[ 0.004431] [00001c02] libusb: debug [linux_get_device_address] getting address for device: 2-1 detached: 0
[ 0.004432] [00001c02] libusb: debug [linux_get_device_address] scan 2-1
[ 0.004439] [00001c02] libusb: debug [linux_get_device_address] bus=2 dev=2
[ 0.004440] [00001c02] libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
[ 0.004441] [00001c02] libusb: debug [linux_enumerate_device] allocating new device for 2/2 (session 514)
[ 0.004451] [00001c02] libusb: debug [linux_get_parent_info] dev 0x56117b65dec0 (2-1) has parent 0x56117b65e2b0 (usb2) port 1
[ 0.004482] [00001c02] libusb: debug [linux_get_device_address] getting address for device: usb3 detached: 0
[ 0.004483] [00001c02] libusb: debug [linux_get_device_address] scan usb3
[ 0.004490] [00001c02] libusb: debug [linux_get_device_address] bus=3 dev=1
[ 0.004491] [00001c02] libusb: debug [linux_enumerate_device] busnum 3 devaddr 1 session_id 769
[ 0.004492] [00001c02] libusb: debug [linux_enumerate_device] allocating new device for 3/1 (session 769)
[ 0.004529] [00001c02] libusb: debug [linux_get_device_address] getting address for device: 3-10 detached: 0
[ 0.004530] [00001c02] libusb: debug [linux_get_device_address] scan 3-10
[ 0.004537] [00001c02] libusb: debug [linux_get_device_address] bus=3 dev=6
[ 0.004538] [00001c02] libusb: debug [linux_enumerate_device] busnum 3 devaddr 6 session_id 774
[ 0.004539] [00001c02] libusb: debug [linux_enumerate_device] allocating new device for 3/6 (session 774)
[ 0.004547] [00001c02] libusb: debug [linux_get_parent_info] dev 0x56117b65e5a0 (3-10) has parent 0x56117b65e6a0 (usb3) port 10
[ 0.004578] [00001c02] libusb: debug [linux_get_device_address] getting address for device: 3-2 detached: 0
[ 0.004579] [00001c02] libusb: debug [linux_get_device_address] scan 3-2
[ 0.004586] [00001c02] libusb: debug [linux_get_device_address] bus=3 dev=2
[ 0.004587] [00001c02] libusb: debug [linux_enumerate_device] busnum 3 devaddr 2 session_id 770
[ 0.004588] [00001c02] libusb: debug [linux_enumerate_device] allocating new device for 3/2 (session 770)
[ 0.004596] [00001c02] libusb: debug [linux_get_parent_info] dev 0x56117b645120 (3-2) has parent 0x56117b65e6a0 (usb3) port 2
[ 0.004627] [00001c02] libusb: debug [linux_get_device_address] getting address for device: 3-3 detached: 0
[ 0.004628] [00001c02] libusb: debug [linux_get_device_address] scan 3-3
[ 0.004635] [00001c02] libusb: debug [linux_get_device_address] bus=3 dev=3
[ 0.004636] [00001c02] libusb: debug [linux_enumerate_device] busnum 3 devaddr 3 session_id 771
[ 0.004637] [00001c02] libusb: debug [linux_enumerate_device] allocating new device for 3/3 (session 771)
[ 0.004645] [00001c02] libusb: debug [linux_get_parent_info] dev 0x56117b645520 (3-3) has parent 0x56117b65e6a0 (usb3) port 3
[ 0.004675] [00001c02] libusb: debug [linux_get_device_address] getting address for device: 3-5 detached: 0
[ 0.004677] [00001c02] libusb: debug [linux_get_device_address] scan 3-5
[ 0.004683] [00001c02] libusb: debug [linux_get_device_address] bus=3 dev=4
[ 0.004685] [00001c02] libusb: debug [linux_enumerate_device] busnum 3 devaddr 4 session_id 772
[ 0.004686] [00001c02] libusb: debug [linux_enumerate_device] allocating new device for 3/4 (session 772)
[ 0.004695] [00001c02] libusb: debug [linux_get_parent_info] dev 0x56117b644130 (3-5) has parent 0x56117b65e6a0 (usb3) port 5
[ 0.004728] [00001c02] libusb: debug [linux_get_device_address] getting address for device: 3-7 detached: 0
[ 0.004730] [00001c02] libusb: debug [linux_get_device_address] scan 3-7
[... cut ...]
[ 0.005339] [00001c02] libusb: debug [libusb_unref_device] destroy device 3.1
[ 0.005341] [00001c02] libusb: debug [libusb_unref_device] destroy device 2.2
[ 0.005342] [00001c02] libusb: debug [libusb_unref_device] destroy device 2.1
[ 0.005344] [00001c02] libusb: debug [libusb_unref_device] destroy device 1.1
[ 0.005461] [00001c03] libusb: debug [linux_udev_event_thread_main] udev event thread exiting
[ 0.005492] [00001c02] libusb: debug [usbi_remove_event_source] remove fd 5
[ 0.005496] [00001c02] libusb: debug [usbi_remove_event_source] remove fd 4
```

gregkh, 2024-09-02:
Good catch, this has been bugging me for a while.  I'll go fix this up now, thanks!

commit c5e98711 (2024-09-02): lsusb: fix memory leak in libusb

When shutting down, tell libusb that we don't want to hold on to any
references anymore.  This fixes the oft-noticed and reported memory leak
when running lsusb.

Fixes #182

Reported-by: mcuee
closed 2024-09-02 by gregkh