gregkh/usbutils #214: usbreset problems

heeplr, 2025-01-20:
The meson.build says
```By default, usbreset does not get installed as it could cause problems, it's in the repo for those that wish to try it out.```

It's not quite clear what kind of problems it can cause. 
Currently distros (namely gentoo) have removed it in 018 and wonder if it's safe to re-add it as an option.

Are we talking about "might or might not work" or more in the direction of "could potentially damage hardware and burn down your house" territory?

gregkh, 2025-01-20:
It could damage your hardware, it could also just flat out not work at all, so you would have to deal with the bug reports from that, as this is not a standard USB requirement of hubs to support this option.

So it's up to you, if you want to run it, great, but we provide no warranty :)

closed 2025-01-20 by gregkh

heeplr, 2025-02-15:
For anyone finding this via search engine: A good (or better) alternative to usbreset is [uhubctl](https://github.com/mvp/uhubctl) if supported by hardware.
It allows to completely cut power to specific ports and thus mimics a physical disconnect/reconnect.