gregkh/usbutils #222: [Question/Bug] usbreset doesn't get installed

RedSnt, 2025-03-11:
I was told usbreset was part of the usbutils package, and I found out that I already had usbutils installed on my nobara (fedora-based) machine, but no usbreset. That's odd I thought. Then I compiled the latest version myself, and sure enough there's the usbreset binary. But when I do sudo meson install it also doesn't install usbreset.  
So now I'm wondering if this is by design or a bug? It's a very useful utility so seems odd it'd be left out.

gregkh, 2025-03-11:
It is by design.  It does not always work, if you want it in the fedora package, and the maintainer there wishes to support it, feel free to ask them to add it to their packaging to add it.

closed 2025-03-11 by gregkh

RedSnt, 2025-03-12:
Thanks, I might just do that. I noticed that Debian and Arch have it included, so I'll probably ask around where I should start that process. Couldn't find an open issue on their bugtracker or forum so maybe it just haven't been reported before.