gregkh/usbutils #223: ContainerID has UNRECOGIZED trailing bytes

ValeZAA, 2025-03-20:
This line is triggered for whatever reason.

https://github.com/gregkh/usbutils/blob/9034a5ffd0d1d65de245923080d2db6d7d4c5337/lsusb.c#L3671

Here is a report from public Internet https://www.reddit.com/r/linux4noobs/s/ATAGegGsx4

The same happens here, i.e. I have realtek hub on the motherboard itself, MSI Carbon Z890 for Arrow Lake.

Then I have this very same ContainerID 20b9cde5-7039-e011-a935-0002a5d5c51b. And in the end some strange UNRECOGNIZED 3 bytes are printed.

```
Container ID Device Capability:
    bLength                20
    bDescriptorType        16
    bDevCapabilityType      4
    bReserved               0
    ContainerID             {20b9cde5-7039-e011-a935-0002a5d5c51b}
  ** UNRECOGNIZED:  03 10 0b
```

Will also add additional data.

ValeZAA, 2025-03-20:
Also some additional problems probably unrelated

dmesg prints this "algorithms for LPM" and indeed if you ask lsusb you will see that iSerial 0000:01:00.2 is     bU1DevExitLat           0 micro seconds  bU2DevExitLat         432 micro seconds which is what triggers LPM warning.

```
[    1.014553] usb usb4: We don't know the algorithms for LPM for this host, disabling LPM.
[    1.014561] usb usb4: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.13
[    1.014562] usb usb4: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.014562] usb usb4: Product: xHCI Host Controller
[    1.014563] usb usb4: Manufacturer: Linux 6.13-rt-amd64 xhci-hcd
[    1.014563] usb usb4: SerialNumber: 0000:01:00.2
[    1.014611] hub 4-0:1.0: USB hub found
[    1.014617] hub 4-0:1.0: 4 ports detected
[    1.014790] xhci_hcd 0000:80:14.0: xHCI Host Controller
[    1.014792] xhci_hcd 0000:80:14.0: new USB bus registered, assigned bus number 5
[    1.015997] xhci_hcd 0000:80:14.0: hcc params 0x20007fc1 hci version 0x120 quirks 0x0000000200009810
[    1.016294] xhci_hcd 0000:80:14.0: xHCI Host Controller
[    1.016295] xhci_hcd 0000:80:14.0: new USB bus registered, assigned bus number 6
[    1.016296] xhci_hcd 0000:80:14.0: Host supports USB 3.2 Enhanced SuperSpeed
```

```

root@kali:~# lsusb -d 1d6b:0003 -vvv

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Negotiated speed: Unknown
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            6.13
  iManufacturer           3 Linux 6.13-rt-amd64 xhci-hcd
  iProduct                2 xHCI Host Controller
  iSerial                 1 0000:00:0d.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
[... 275 lines cut ...]
  wHubDelay             0 nano seconds
  DeviceRemovable    0xa8 0x00
 Hub Port Status:
   Port 1: 0000.02a0 5Gbps power Rx.Detect
   Port 2: 0000.02a0 5Gbps power Rx.Detect
   Port 3: 0000.0263 5Gbps power suspend enable connect
     Ext Status: 0000.0055
       RX Speed Attribute ID: 5 Lanes: 1
       TX Speed Attribute ID: 5 Lanes: 1
   Port 4: 0000.02a0 5Gbps power Rx.Detect
   Port 5: 0000.0263 5Gbps power suspend enable connect
     Ext Status: 0000.0055
       RX Speed Attribute ID: 5 Lanes: 1
       TX Speed Attribute ID: 5 Lanes: 1
   Port 6: 0000.02a0 5Gbps power Rx.Detect
   Port 7: 0000.0263 5Gbps power suspend enable connect
     Ext Status: 0000.0044
       RX Speed Attribute ID: 4 Lanes: 1
       TX Speed Attribute ID: 4 Lanes: 1
   Port 8: 0000.02a0 5Gbps power Rx.Detect
   Port 9: 0000.02a0 5Gbps power Rx.Detect
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

ValeZAA, 2025-03-20:
Oh, and this also happens:

[    1.018098] usb usb6: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.13
[    1.018099] usb usb6: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.018100] usb usb6: Product: xHCI Host Controller
[    1.018100] usb usb6: Manufacturer: Linux 6.13-rt-amd64 xhci-hcd
[    1.018100] usb usb6: SerialNumber: 0000:80:14.0
[    1.018151] hub 6-0:1.0: USB hub found
[    1.018170] hub 6-0:1.0: 9 ports detected
[    1.018815] **usb: port power management may be unreliable**

This is issue from here: https://lore.kernel.org/all/7efc3c5b-3237-30c9-1ff2-88747897ed57@linux.intel.com/

ValeZAA, 2025-03-20:
Oh, https://chromium.googlesource.com/chromiumos/platform/ec/+/master/include/usb_descriptor.h#84

This is type 0b. PCSTIME? Not defined in here

https://github.com/gregkh/usbutils/blob/9034a5ffd0d1d65de245923080d2db6d7d4c5337/lsusb.c#L56

gregkh, 2025-03-29:
Patches to add support for this extension to usbutils are gladly appreciated.  As I don't have this hardware, I can't do so myself at the moment, sorry.

ValeZAA, 2025-03-29:
As far as I understand it is just 0 byte extension (well 3 byte with metadata), where 03 10 0b, first byte is length, 10 means bDescriptorType and 0b means bDevCapabilityType PCSTIME.

Unless there are more to parse in other parts of the lsusb -vv?

gregkh, 2025-10-02:
Again, I'll gladly take patches for this, but as I can't test it it's hard for me to be able to implement it at the moment, sorry.