gregkh/usbutils #226: Reproducibly unable to acquire information about certain devices.

RokeJulianLockhart, 2025-04-09:
#### Example

When I invoke `lsusb -v`, I reproducibly see:

<blockquote>

~~~log
Couldn't get device capability descriptors
can't get hub descriptor, LIBUSB_ERROR_TIMEOUT (Resource temporarily unavailable)
cannot read device status, Resource temporarily unavailable (11)
~~~

</blockquote>

An example is:

<details>

<blockquote>

~~~YAML
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         2 TT per port
  bMaxPacketSize0        64
  idVendor           0x05e3 Genesys Logic, Inc.
  idProduct          0x0610 Hub
  bcdDevice           34.04
  iManufacturer           1 GenesysLogic
  iProduct                2 USB2.1 Hub
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      1 Single TT
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      2 TT per port
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0049
  bNumDeviceCaps          5
Couldn't get device capability descriptors
can't get hub descriptor, LIBUSB_ERROR_TIMEOUT (Resource temporarily unavailable)
cannot read device status, Resource temporarily unavailable (11)
~~~

</blockquote>

</details>

[^2]

[^2]: [`unix.stackexchange.com/revisions/793511/1`](https://unix.stackexchange.com/revisions/793511/1#:~:text=Couldn't%20get%20device%20capability%20descriptors,status%2C%20Resource%20temporarily%20unavailable%20(11))

#### Corroboration

I've located rare occurrences of this elsewhere:

1. [`reddit.com/r/dietpi/comments/v9iuw7/comment/mm6ztkb`](https://www.reddit.com/r/dietpi/comments/v9iuw7/comment/mm6ztkb/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button)

1. [`unix.stackexchange.com/revisions/572008/3`](https://unix.stackexchange.com/revisions/572008/3#:~:text=can't%20get%20hub%20descriptor%2C%20LIBUSB_ERROR_TIMEOUT,debug%20descriptor:%20Resource%20temporarily%20unavailable)

1. [`unix.stackexchange.com/revisions/360125/4`](https://unix.stackexchange.com/revisions/360125/4#:~:text=2:%200000.0100%20power-,can't%20get%20debug%20descriptor:%20Resource%20temporarily%20unavailable,-Device%20Status:%20%20%20%20%200x0001)

#### Diagnosis

I see that it's mentioned at the undermentioned, [^1] if of use:

https://github.com/gregkh/usbutils/blob/9034a5ffd0d1d65de245923080d2db6d7d4c5337/lsusb.c#L3627

[^1]: [`9034a5ffd0d1d65de245923080d2db6d7d4c5337/lsusb.c#L3627C20-L3627C62`](https://github.com/gregkh/usbutils/blob/9034a5ffd0d1d65de245923080d2db6d7d4c5337/lsusb.c#L3627C20-L3627C62)

#### Environment

[`usbutils-017-3.fc41.x86_64`](https://koji.fedoraproject.org/koji/rpminfo?rpmID=39457104#:~:text=Information%20for%20RPM-,usbutils%2D017%2D3.fc41.x86_64.rpm,-ID%0939457104)

gregkh, 2025-04-09:
What is the issue here?  You do not have permissions to access that information when running that command.  If you elevate your permission, that message will go away, right?

RokeJulianLockhart, 2025-04-09:
> https://github.com/gregkh/usbutils/issues/226#issuecomment-2789502899

@gregkh, apologies – I meant that it appears when invoked with `sudo`. It actually doesn't appear without `sudo`, from my limited evaluations. I merely forgot to prepend it when I wrote the issue.

gregkh, 2025-04-09:
I do not know, sorry, looks like something else is bound to that hub device and will not allow `lsusb` to poke at it.

What else is running on the system at the same time?  If you boot to only a console, does it work there?

RokeJulianLockhart, 2025-04-09:
>  If you boot to only a console, does it work there?

@gregkh, I can't *boot* to a console easily, because I have SDDM installed by default with my DE. However, I can boot, log out, then, at SDDM's GUI, switch (with <kbd>Control</kbd> + <kbd>Alt</kbd> + <kbd>F(n)</kbd>) to a TTY CLI. When I do so, the behaviour is identical.

> What else is running on the system at the same time?

I'll include it in an edit.

<blockquote>

~~~Markdown

~~~

</blockquote>

ValeZAA, 2025-04-09:
This is a known issue. The only way to fix this is to remove driver and read data and then reattach. But that will break devices. See patches: https://github.com/gregkh/usbutils/pull/52

ValeZAA, 2025-04-09:
There is another way to read some of the descriptors that are not that sensitive (for example HID is not sensitive)... 

cat /sys/kernel/debug/hid/*/rdesc

But naturally some devices are real time and cannot be manipulated without some serious hacks.
if you try lsusb -vvv there will be even more

can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable

I will find my notes from years ago and update this post

closed 2025-10-02 by gregkh

RokeJulianLockhart, 2025-10-02:
> https://github.com/gregkh/usbutils/issues/226#event-20061073942

@gregkh, perhaps, re-close as unplanned.